arancormonk/dsd-neo #1: Dsd-fme TETRA mode

joannesberndt, 2025-09-14:
Where did your fork version of DSD-FME disappear? Could you add TETRA mode decoding?

romanremus, 2025-09-17:
Hello arancormonk, and congratulations for the code, it is superb, I have the latest version but in the Ncurses terminal SNR does not look exactly like it does in the code, do you have any idea? I have attached a picture

romanremus, 2025-09-17:
[image]

romanremus, 2025-09-17:
I think this symbol would work best with SNR, displaying a level, what do you think?

romanremus, 2025-09-17:
[image]

romanremus, 2025-09-18:
I honestly don't understand how it works or how it displays the SNR, shouldn't it increase from a small line to the largest line as a function of the signal strength when it receives a signal?

arancormonk, 2025-09-20:
> Hello, Have you started adding TETRA code in dsd-neo? because I look at the source code and I don't see anything.

there's no TETRA for 4000+ miles from where i am... so it'll be a long term goal. there is already good work being done in other open source projects, so when i have the free time and RF captures of TETRA i can start to work on it.

M810192, 2025-11-01:
I can receive two different tetra networks, so I could provide you with captures. Just tell me what you need and I will try to get it

Saqibjaved25, 2026-01-07:
I downloaded window version and run the exe, dsd-neo.exe, then set all the required things but when i transmit the signal from my DMR, the software doesn't show anything nor any error or something.

[image]

arancormonk, 2026-01-08:
@Saqibjaved25  yea, that ppm is extreme! 100!? can you use sdr# or sdrpp first to determine what the sane values of your rtl dongle should be? use a known good carrier e.g. NOAA if you're in the US.

also, i'd recommend starting a new conversation as a github discussion so as to not continue polluting this issue: https://github.com/arancormonk/dsd-neo/discussions... this has nothing to do with TETRA

haoyunZhou, 2026-09-21:
I've integrated TETRA support into my fork and spent quite a bit of time working on it. I'm not sure about the best way to submit these changes upstream. Would you prefer a single pull request, or should I split the TETRA implementation into several smaller PRs?

romanremus, 2026-09-21:
haoyunZhou :  What do you think about the TETRA encryption methods? Could they be added ? 
https://github.com/MidnightBlueLabs/TETRA_crypto
https://github.com/MidnightBlueLabs/TETRA_etee
https://github.com/osmocom/osmo-tetra

romanremus, 2026-09-21:
It decodes in real time; how do you start it for TETRA decoding?

arancormonk, 2026-09-21:
> I've integrated TETRA support into my fork and spent quite a bit of time working on it. I'm not sure about the best way to submit these changes upstream. Would you prefer a single pull request, or should I split the TETRA implementation into several smaller PRs?

Wow, sure! Smaller PRs would be much easier for me to review if it isn't too much trouble, I'll leave it at your discretion.

romanremus, 2026-09-21:
Hi, I tried using this command, but the bandwidth shows 48 instead of 72, and it’s not decoding anything; am I doing something wrong during startup? 
dsd-neo.exe -fT -i rtl:0:390.163M:10:-5:72:0:2 -o pulse -N -Z

ysamouhos, 2026-09-21:
Is there a branch we can test ?

romanremus, 2026-09-21:
Hi, try https://github.com/haoyunZhou/dsd-neo/tree/main

ysamouhos, 2026-09-21:
same errors
Am I missing something ?

romanremus, 2026-09-21:
I managed to compile the code using Visual Studio, but I had to modify it to support BW72; I also created a batch file in the build folder where I set a frequency, and it seems to be decoding, though I'm not seeing any messages in the terminal.

ysamouhos, 2026-09-21:
Well I am running in Linux ubuntu noble (24.04.4)

ysamouhos, 2026-09-21:
Hmm it does not decode anything. (I mean unencrypted voice)

romanremus, 2026-09-21:
Well, I don't know what to say; it decodes for me, but nothing shows up in the Ncurses terminal. As for the voice, I think all the networks here are encrypted; the issue I had with decoding was that it only accepted a maximum bandwidth of 48, not 72.

haoyunZhou, 2026-09-22:
I’ve pushed a few more changes, including several fixes to the TETRA voice path, RTL bandwidth handling, and demod configuration.
TETRA voice (dbfceb8d). An NTS1 burst is treated as a single 432-bit logical channel. Block 1 and Block 2 are now descrambled using one continuous LFSR run. If the block passes the SCH/F CRC check (interleaver a=103, rate 2/3), it is treated as signalling and is not passed to the speech vocoder. Only blocks that fail the CRC check are treated as TCH/FS.

NTS2 is not treated as speech. It is handled as an inverted NTS1 alias, while polarity is still determined solely from the synchronization burst.

Block 1 consists of the 108 dibits at symbol ages 18..125, following the 7-dibit AACH half.

The ETSI ACELP codec is not included in the repository and is not built by the top-level CMake project. tools/tetra/acelp_adapter is provided as a separate utility. Developers can point -DETSI_TETRA_C_CODE to their local ETSI C-CODE directory and select the resulting adapter through TETRA_VOCODER_CMD.

On Linux and macOS, the command is launched using sh -c; on Windows, it is launched using cmd.exe /C. The pipe protocol sends 137 bytes per speech frame to the vocoder and receives 480 bytes of PCM16LE audio.

RTL (0fc02781). Engine startup now accepts a 72 kHz RTL bandwidth. Previously, the fallback was limited to 48 kHz, causing configurations such as rtl:...:72:... to be rejected.

Config (4942176f). An explicit demodulator lock in auto mode is now preserved only when the saved configuration is actually dPMR. The previous check relied on opts->frame_dpmr, which could incorrectly force dPMR demodulation when another mode was selected.

romanremus, 2026-09-22:
HaoyunZhou : Is it possible to have all the TETRA decoding messages displayed in the dsd-neo Terminal (-N)? What are your thoughts on the encryption codes? Thanks a lot for the TETRA support in dsd-neo, though I think all the networks here are encrypted.

ysamouhos, 2026-09-23:
@haoyunZhou Please add support for linux of how we should build the acelp coded and how the chain should run

ysamouhos, 2026-09-23:
...for example filesnames in linux ARE case snesitive. 

dsd-neo/third_party/tetra_acelp/C-CODE/SDEC_TET.C:18:10: fatal error: source.h: No such file or directory
   18 | #include "source.h"

ysamouhos, 2026-09-23:
Anyway still not able to hear any unencrypted voice.. 

pipeline is sending 48Khz iq to localhost tcp:41439 and expects voice output to udp:40420 
dsd-neo -i tcp:127.0.0.1:41439 -o udp:127.0.0.1:40420 -fT -Z

Even though env variable is declare -x TETRA_VOCODER_CMD="/usr/local/bin/tetra-acelp-adapter" I do not see it running at anypoint.

romanremus, 2026-09-25:
HaoyunZhou : Please consider adding support for encryption; you can use the code from MidnightBlue Tetra or Osmocom-Tetra as a reference. It would also be very helpful if the encryption type (TEA 1, 2, 3, etc.) could be displayed. Thanks

haoyunZhou, 2026-09-26:
I committed TETRA fixes. All five focused tests pass

haoyunZhou, 2026-09-26:
[romanremus](https://github.com/romanremus) :  I can give it a try and see how it goes.

ysamouhos, 2026-09-26:
let me test again as well

ysamouhos, 2026-09-26:
@haoyunZhou do we use the tetra-live-voice branch ?

M810192, 2026-09-28:
as others mentioned, it would be great to add support for decryption with a known key. 
midnight blue has the basic code, but also some work has already been done here: https://github.com/sq5bpf/osmo-tetra-sq5bpf-2

Saqibjaved25, 2026-10-09:
I have recorded some **TETRA IQ** files containing **voice data**, and they are not encrypted. I understand that you are working on this area, so I thought these files might be useful for your **research** and **development**. If you are interested, I would be happy to provide them.

haoyunZhou, 2026-10-12:
@Saqibjaved25 Thanks for reaching out. I’ve added the two modes (TMO Cluster Mode and DMO Direct Mode) in my latest changes. Please wait a bit, I’ll push them shortly.

Saqibjaved25, 2026-10-12:
@haoyunZhou  Can I know whether **TETRA mode** works only with RTL-SDR signals, or does it also support separate commands such as **-fd, -i tcp:127.0.0.1:0000, -fx, and -fs**?
and just like other modes we will be able to hear voice data.

romanremus, 2026-10-12:
HaoyunZhou : I’ve found a few more things that might be useful to you, and I can also provide samples of encrypted TETRA signals; I think it would be a great help if, by analyzing the signaling, we could determine the type of encryption being used—TEA 1, TEA 2, or TEA 3. Thank you.
https://github.com/Steiynbrodt/sdrpp-tetra-demodulator-decryptionsupport-

romanremus, 2026-10-14:
HaoyunZhou : Could you also add decryption using the reduced key from TEA1 to the code? It would be very helpful if you could make it display the type of encryption used on a network

haoyunZhou, 2026-10-14:
@Saqibjaved25 @romanremus TMO Cluster Mode and DMO Direct Mode both ok ,  tetra-live-voice branch

Saqibjaved25, 2026-10-14:
@haoyunZhou That’s absolutely excellent! I’d love to test it once it’s ready. There are quite a lot of TETRA signals in the air in my area, so I’d be happy to help with testing and provide feedback.

romanremus, 2026-10-14:
Saqibjaved25 : Your Tetra Signals is TEA2 ?

Saqibjaved25, 2026-10-14:
@romanremus Most of the signals I have are TEA1. However, if you need TEA2 or TEA3 signals, I can provide those as well. That should be possible for me.

romanremus, 2026-10-14:
Saqibjaved25 : That’s exactly the problem for me—I don’t know what type of encryption it is, though it would really help to find out. How did you figure out which encryption yours uses?

Saqibjaved25, 2026-10-14:
@romanremus If you know the frequency ranges used by TEA1, TEA2, and TEA3, you might be able to identify or make an educated guess about the encryption algorithm used in recordings. Perhaps you could investigate this on your end. on which frequency did you captured the signal?

romanremus, 2026-10-14:
In my area, I only get signals in the 390–395 MHz range; I believe they are TEA2 units—which I’ve heard can't be modified—but if you have a TEA1, check out SQ5BPF's code on GitHub; he has something interesting for the TEA1

haoyunZhou, 2026-10-14:
The bandwidth must be 72 kHz.

Saqibjaved25, 2026-10-14:
@haoyunZhou @romanremus here is a file which contains some signals, you can check it out as well. https://drive.google.com/file/d/1bvbeWtKO0FpcHjy0rMRHGLXVY9PgrIGL/view?usp=sharing

arancormonk, 2026-10-15:
i've made a metric buttload of changes and additions to the tetra PR #585 in an attempt to align the protocol addition with the existing dsd-neo architecture and codebase rules... once the final checks pass, i'm going to merge it.

i'd highly recommend creating separate single scoped issues instead of continuing to use this very long issue thread (there are 60 comments). that will help me and @haoyunZhou address each item moving forward after this addition.

be aware, i had to make some fundamental adjustments to the original PR approach due to obvious licensing issues... the way it is now should be fine, though. re-read the docs (or use an AI to figure it out) for the latest usage guidelines.

closed 2026-10-15 by arancormonk

Attilah76a, 2026-10-26:
> 29\. [@haoyunZhou](https://github.com/haoyunZhou) [@romanremus](https://github.com/romanremus) here is a file which contains some signals, you can check it out as well. https://drive.google.com/file/d/1bvbeWtKO0FpcHjy0rMRHGLXVY9PgrIGL/view?usp=sharing

you still have some signals?