arancormonk/dsd-neo #3: How to  Enter the decryption key for vertex Standard  Enhanced Encryption

taradeved, 2025-09-28:
help me!

dsd-neo  -1 bbbbbbbbb It does not decrypt
help me!
It does not recognize the encryption type.
It would be good to add detection of the type of encoding used in the vertex because dsd-neo recognizes all vertex samples as Bp.

arancormonk, 2025-09-28:
as far as i'm aware, Vertex “enhanced” encryption isn’t supported by upstream DSD‑FME... it's something we'll have to add since Vertex ([they are part of Motorola fwiw](https://www.motorolasolutions.com/en_us/products/vertex-standard.html) ) obviously doesn't want to follow the same implementation, at least from what i can tell in the work already done by lwvmobile... and it's probably on purpose.
 
i can work on adding support, but to finish the keystream analysis I need aligned clear/encrypted pairs and known plaintext.

  - What To Capture (**exactly 4 items**)
      - Voice (clear): one capture of a short voice transmission unencrypted.
      - Voice (encrypted): the exact same voice transmission, but encrypted.
      - Text (clear): one capture of a short text message unencrypted.
      - Text (encrypted): the exact same text message, but encrypted.
  - Recording Guidelines (**must follow exactly**)
      - Same RF parameters for each pair: same frequency, Color Code, timeslot, talkgroup, and radio ID.
      - Same content:
          - Voice: use the same audio source, **preferably a pre‑recorded test phrase played twice** (once clear, once encrypted), not a human re‑speaking. Add 1–2 seconds of “quiet/open mic” before and after the phrase both times. Keep
  duration ~10–20 seconds.
          - Text: use the same exact message bytes (e.g., “00000000000000”) for both clear and encrypted sends.
      - Same keying: **use the same key for all encrypted samples**. If your software shows a key ID or algorithm name, note it.
      - Don’t trim or post‑process: provide raw captures as recorded.
  - File Formats (either of these works; both is even better)
      - Preferred: DMR symbol capture .bin (OP25/FME-style dibit file) — one file per item.
      - Acceptable: 48 kHz mono WAV discriminator/baseband audio — one file per item.
  - Helpful Metadata (include in your message back)
      - Key value used for encryption (as entered), and any key ID displayed on the radio.
      - Radio brand/model and programming options used for “enhanced” encryption.
      - Talkgroup ID, timeslot, Color Code, and whether this was simplex or via repeater.
      - For the text message: confirm the exact plaintext string and character encoding if non‑ASCII (spaces, punctuation, case, etc.).
      - Timestamps or ordering so I can pair clear↔encrypted samples correctly.
  - Tips To Ensure Samples Are Truly “Identical”
      - Voice: **use the same pre‑recorded WAV/MP3 played twice into the radio, with the same start/stop timing. Avoid live re‑speaking**.
      - If your radio supports **sending a stored voice memo or test prompt, use that for exact repeatability**.
      - Keep transmissions back‑to‑back to minimize any system changes.
  - Scope/Size
      - 10–20 seconds per voice capture is enough. One short text message per clear/encrypted is fine.
      - Please avoid compressing the files (ZIP is OK; no lossy audio re‑encoding).

taradeved, 2025-09-29:
> as far as i'm aware, Vertex “enhanced” encryption isn’t supported by upstream DSD‑FME... it's something we'll have to add since Vertex ([they are part of Motorola [...]

I will work on this and then send it to you.

taradeved, 2026-05-28:
I tried this option "dsd-neo --dmr-vertex-ks-csv vertex_key.csv" but it doesn't work with Vertex device
and I didn't hear any sound even though the key_hex and keystream are correct
but this option  "dsd-neo -S 49:F5C608EB778A0000" works and hear the sound clearly
Please help me solve this matter.

arancormonk, 2026-05-28:
> I tried this option "dsd-neo --dmr-vertex-ks-csv vertex_key.csv" but it doesn't work with Vertex device and I didn't hear any sound even though the key_hex and keystream are correct but this option "dsd-neo -S 49:F5C608EB778A0000" works and hear the sound clearly Please help me solve this matter.

`-S 49:F5C608EB778A0000` force-applies that keystream directly, but `--dmr-vertex-ks-csv` only applies when the decoded Vertex ALG/key metadata matches a CSV row.

Please try running it with the key loaded too:

```sh
dsd-neo -1 1234567891 --dmr-vertex-ks-csv vertex_key.csv ...
```
Also please attach or paste vertex_key.csv and the stderr/log lines around the Vertex/PI messages. I need to confirm whether the CSV is loading and whether the OTA key/ALG metadata matches the CSV key.

taradeved, 2026-05-29:
This is  unencrypted text capture/log for the same message 
 UDP Protocol; Datagram Len: 46; UDP Checksum: 106F; Unknown UDP Port;
 Slot 1 - Multi Block PDU Message
  4500004200000000401162A9
  0C0000010C000002138F138F
  002E106F0E00020000000000
  000030003000300030003000
  300030003000300030003000
  300030003000000000000000
  000000007A372216

taradeved, 2026-05-29:
dsd-neo   -i tcp -1 000413945B --dmr-vertex-ks-csv vertex_key.csv  -Z
NOTICE: Loaded 1 Vertex key->keystream mappings from 'vertex_key.csv'.
NOTICE: RC4/DES encryption key loaded: [redacted]
NOTICE: Logging MBE/PDU payloads to console.
NOTICE: ------------------------------------------------------------------------------
NOTICE: | Digital Speech Decoder: DSD-neo v1.0.0 (unknown) 
NOTICE: ------------------------------------------------------------------------------
NOTICE: MBElib-neo Version: 2.0.0
NOTICE: TCP Direct Link: NOTICE: localhost:NOTICE: 7355 
NOTICE: TCP Connection Success!
Audio In Device: tcp
02:53:33 Sync: +DMR MS/DM MODE/MONO | Color Code=XX | VLC  \n SLOT ? Protected LC  FLCO=0x1E FID=0x68  (CRC ERR) \n DMR PDU Payload [DE][68][50][04][06][C1][D0][11][46][E0][DE][02]                                                                                          
02:53:33 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VLC  \n SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02][F2][DF][EC]                                                                        
02:53:33 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VLC  \n SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02][F2][DF][EC]                                                                        

[DMR SM] voice-sync
02:53:33 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VC* 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 

[DMR SM] voice-sync
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0D27B8BCD38000 err = [0] [0] 
 AMBE 0D27B8BAD39200 err = [0] [0] 
 AMBE 0D27B8BCD30280 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0D26782CD3E280 err = [0] [0] 
 AMBE 0D25182CF3E380 err = [0] [0] 
 AMBE 9F1A5D12F06F00 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 1D07E06D108200 err = [0] [0] 
 AMBE 0D247BC2FBE880 err = [0] [0] 
 AMBE 0D2FABC4DBFB80 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0D23F8D4FBD280 err = [0] [0] 
 AMBE 0D33FBD5F3F280 err = [0] [0] 
 AMBE AD21F83E12F200 err = [0] [0] 
 SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02] SB: 00000000000 - 000;                                                                                                                          

[DMR SM] voice-sync
02:53:33 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VC* 
 AMBE 7D1CEBE3769E80 err = [0] [0] 
 AMBE 0D273CCDBB1080 err = [0] [0] 
 AMBE D915AD12E6E980 err = [0] [0] 

[DMR SM] voice-sync
 AMBE D925F965AC5980 err = [0] [0] 
 AMBE 0D26943C336A80 err = [0] [0] 
 AMBE 99248D1C4CF400 err = [0] [0] 
..............................................................................
dsd-neo   -i tcp  --dmr-vertex-ks-csv vertex_key.csv  -Z
NOTICE: Loaded 1 Vertex key->keystream mappings from 'vertex_key.csv'.
NOTICE: Logging MBE/PDU payloads to console.
NOTICE: ------------------------------------------------------------------------------
NOTICE: | Digital Speech Decoder: DSD-neo v1.0.0 (unknown) 
NOTICE: ------------------------------------------------------------------------------
NOTICE: MBElib-neo Version: 2.0.0
NOTICE: TCP Direct Link: NOTICE: localhost:NOTICE: 7355 
NOTICE: TCP Connection Success!
Audio In Device: tcp
SPS hunt: trying 8 sps (sym=6000, demod=48000)
02:55:01 Sync: -YSF  FN: 10/10 FICH (CRC ERR)  FICH: [00][00][4D][F7]DCH (CRC ERR) 
 DCH1: [B5][60][0F][01][99][DE][70][A2][3E][8F][E1][FC][7E][96][D8][10][9C][6A][FC][C9][5A][D3]DCH (CRC ERR) 
 DCH1: [92][68][96][59][4B][B5][D7][5B][A0][76][5C][5B][2C][02][12][51][20][B3][9E][3C][B9][AD]
Sync: no sync
SPS hunt: trying 10 sps (sym=4800, demod=48000)
02:55:03 Sync: +DMR MS/DM MODE/MONO | Color Code=XX | VLC  \n SLOT ? Protected LC  \n SLOT ? Unknown LC  FLCO=0x30 FID=0xE8  (CRC ERR) \n DMR PDU Payload [F0][E8][11][84][0E][81][F0][41][41][C0][4A][66]
02:55:03 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VLC  \n SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02][F2][DF][EC]                                                                        
02:55:03 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VLC  \n SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02][F2][DF][EC]                                                                        

[DMR SM] voice-sync
02:55:03 Sync: +DMR MS/DM MODE/MONO | Color Code=01 | VC* 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 

[DMR SM] voice-sync
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 
 AMBE F801A99F8CE080 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0DCF3972FBB980 err = [0] [0] 
 AMBE 0DDF39A3F3F380 err = [0] [0] 
 AMBE 0DCF3823DBB200 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0DCF38A2D3B000 err = [0] [0] 
 AMBE 0DBF11F393F800 err = [0] [0] 
 AMBE 4D271C3C1FDE00 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 3D339C1517EE00 err = [0] [0] 
 AMBE 2D1D9124FF3A80 err = [0] [0] 
 AMBE 0D1FC8C4FBD880 err = [0] [0] 

[DMR SM] voice-sync
 AMBE 0D25D8B4F3C280 err = [0] [0] 
 AMBE 3F130D48EE6480 err = [0] [0] 
 AMBE 8D4910C53B2180 err = [0] [0] 
 SLOT ? TGT=1 SRC=2 FLCO=0x00 FID=0x10 SVC=0x40 Group Encrypted Call  \n DMR PDU Payload [00][10][40][00][00][01][00][00][02] SB: 00000000000 - 000;